xArm-Developer/xArm-Python-SDK #139: Potential issue with dc control box and joint range check

IrvingF7, 2025-04-17:
Hi! Thanks for open-sourcing this.

I use a xArm 6 with a DC control box, and I noticed that I cannot set servo angles.

The issue is this [line](https://github.com/xArm-Developer/xArm-Python-SDK/blob/b6b60700bdbcea7e397ad9f2dc4ce3365e3876ed/xarm/x3/xarm.py#L71)

```
device_type = int('{}1305'.format(self.axis)) if self.sn and int(self.sn[2:6]) >= 1305 and int(self.sn[2:6]) < 8500 else self.device_type
```

Basically, maybe because I am using a DC control box, my robot doesn't seem to have a sn in the eyes of this SDK, so `self.sn` would be an empty string, and would cause an error when converted to an int.

In most other places, `self.sn` would be checked against `isdigit` (I will find some examples later), which leads me to believe this is a small bug. I made some modifications to let `device_type = self.device_type directly, and after that servo angle works fine

But maybe I am missing something or doing something wrong. In that case, I would really appreciate it if you can let me know

Cheers

penglongxiang, 2025-04-17:
Hi, Could you please provide the actual Serial Number attached at the robot base? And also please download the log package through Studio at Settings->My Device->log->Download, you may sent it to jason@ufactory.cc. We will help diagnose the cause.

IrvingF7, 2025-04-18:
> Hi, Could you please provide the actual Serial Number attached at the robot base? And also please download the log package through Studio at Settings->My Device->log->Download, you may sent it to [jason@ufactory.cc](mailto:jason@ufactory.cc). We will help diagnose the cause.

Thanks for the prompt reply. I will do that when I got back to the lab. Meanwhile, if you look at this [line](https://github.com/xArm-Developer/xArm-Python-SDK/blob/b6b60700bdbcea7e397ad9f2dc4ce3365e3876ed/xarm/x3/base.py#L422) in `x3/base.py`, it does a very similar thing but says

`self._arm_type_is_1300 = int(xarm_sn[2:6]) >= 1300 if xarm_sn[2:6].isdigit() else False`, so I just did something similar to make sure sn is not an empty string, and was able to by pass this error. Is the original code in `x3/xarm.py` not a bug?

penglongxiang, 2025-04-24:
Hi, we have to check the log to see why the `sn` extracted by SDK is empty (this seems to be the major issue) and also help us to confirm if a logic bug exist or not. Bypassing the error may not be a safe workaround. Please provide the log to us first to better confirm the cause. Thanks.